adafruit/nina-fw #6: Problems uploading to metro_ m4_airlift

jerryneedell, 2019-06-17:
See comments in this PR.

https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/45#issuecomment-502840266

jerryneedell, 2019-06-17:
I reproduced the same issue  loading the same file to a PyPortal. Failed at the sme percentage loaded. (26%)
```
The same workaround was also successful on the Pyportal
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ /usr/local/bin/esptool.py  --port /dev/ttyACM0 --before no_reset --baud 115200 write_flash 0 NINA_W102-1.3.1.bin 
esptool.py v2.5.1
Serial port /dev/ttyACM0
Connecting.......
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: c4:4f:33:0e:04:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1129472 bytes to 610164...
Writing at 0x00024000... (26 %)
A fatal error occurred: Timed out waiting for packet header
```
reload with separate files
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ python /home/jerryneedell/projects/esp-idf-3.2/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyACM0 --baud 115200 --before no_reset  write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/bootloader/bootloader.bin 0x30000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/nina-fw.bin 0x8000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/partitions.bin
esptool.py v2.6
Serial port /dev/ttyACM0
Connecting.......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: c4:4f:33:0e:04:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 15168 bytes to 9930...
Wrote 15168 bytes (9930 compressed) at 0x00001000 in 3.3 seconds (effective 37.2 kbit/s)...
Hash of data verified.
Compressed 932672 bytes to 564789...
Wrote 932672 bytes (564789 compressed) at 0x00030000 in 185.6 seconds (effective 40.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 120...
Wrote 3072 bytes (120 compressed) at 0x00008000 in 0.1 seconds (effective 470.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
Note -- I have been using both esptool.py 2.5.1 and 2.6  -- both behave the same
```jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ /usr/local/bin/esptool.py --chip esp32 --port /dev/ttyACM0 --baud 115200 --before no_reset  write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/bootloader/bootloader.bin 0x30000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/nina-fw.bin 0x8000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/partitions.bin
esptool.py v2.5.1
Serial port /dev/ttyACM0
Connecting........_____...
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: c4:4f:33:0e:04:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 15168 bytes to 9930...
Wrote 15168 bytes (9930 compressed) at 0x00001000 in 3.3 seconds (effective 37.2 kbit/s)...
Hash of data verified.
Compressed 932672 bytes to 564789...
Wrote 932672 bytes (564789 compressed) at 0x00030000 in 185.6 seconds (effective 40.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 120...
Wrote 3072 bytes (120 compressed) at 0x00008000 in 0.1 seconds (effective 468.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

jerryneedell, 2019-06-18:
one more piece of information -- I am able to upload the same single file (adafruit repo build) to a feather_huzzah_esp32 via the USP port
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ /usr/local/bin/esptool.py  --port /dev/ttyUSB0  --baud 115200 write_flash 0 NINA_W102-1.3.1.bin 
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: bc:dd:c2:f7:fe:29
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1129472 bytes to 610164...
Wrote 1129472 bytes (610164 compressed) at 0x00000000 in 53.8 seconds (effective 168.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

jerryneedell, 2019-06-18:
as requested by @nevercast  in the original discussion in the ESP32SPI PR I reran both load attemps with the --no-stub option enabled
the single file still fails -- at a higher percentage
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ cp ~/Downloads/Metro_M4_WiFi_ESP32_Passthru.UF2 /media/jerryneedell/METROM4BOOT/
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ /usr/local/bin/esptool.py  --port /dev/ttyACM0 --before no_reset --no-stub --baud 115200 write_flash 0 NINA_W102-1.3.1.bin 
esptool.py v2.5.1
Serial port /dev/ttyACM0
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: c4:4f:33:0e:a9:dd
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Took 3.66s to erase flash block
Writing at 0x0005e400... (34 %)
A fatal error occurred: Timed out waiting for packet header
```
reload with multi-file
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/nina-fw$ /usr/local/bin/esptool.py --chip esp32 --port /dev/ttyACM0 --baud 115200 --before no_reset  --no-stub write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/bootloader/bootloader.bin 0x30000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/nina-fw.bin 0x8000 /home/jerryneedell/projects/adafruit_github/nina-fw/build/partitions.bin
esptool.py v2.5.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: c4:4f:33:0e:a9:dd
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Flash params set to 0x0220
Compressed 15168 bytes to 9930...
Took 0.17s to erase flash block
Wrote 15168 bytes (9930 compressed) at 0x00001000 in 1.0 seconds (effective 118.2 kbit/s)...
Hash of data verified.
Erasing flash...
Compressed 932672 bytes to 564789...
Took 2.74s to erase flash block
Wrote 932672 bytes (564789 compressed) at 0x00030000 in 55.7 seconds (effective 134.0 kbit/s)...
Hash of data verified.
Erasing flash...
Compressed 3072 bytes to 120...
Took 0.06s to erase flash block
Wrote 3072 bytes (120 compressed) at 0x00008000 in 0.1 seconds (effective 225.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

jerryneedell, 2019-06-18:
@nevercast  -- I was puzzled by your comment
"Both images are about the same size, so they should behave similar. A big change is the starting address.
"

the two images -- one built in the adafruit repo and one built in the docmollo repo have similar sizes and the same load addresses. But one loads and the other fails when the single file load is attempted.

docmollo, 2019-06-18:
I do know that while I was working on this code, uploading to the ESP32 via the passthrough was more finicky than I would like. That said, I was able to get things to work fairly consistently. I'll take a look at this after I've got the ESP32SPI PR closed, so it might be a week or so before I have time to look at this issue. Thanks for helping test the code!

brentru, 2019-10-08:
@jerryneedell  We have a guide for upgrading the ESP32 firmware (here: https://learn.adafruit.com/upgrading-esp32-firmware). I've been uploading/flashing nina-fw the past two weeks without running into a timeout.

Is it ok to close this issue and re-open in the future if we encounter this again?

jerryneedell, 2019-10-08:
Fine with me -- closing

closed 2019-10-08 by jerryneedell