adafruit/nina-fw #10: read from closed socket

chegewara, 2019-07-29:
Hi, at beginning i want to apologize if i dont know what i am saying, because i dont know sockets and LWIP.

Here is problem i am facing with:
- if in my app before i read from socket i check if any data is available with `availDataTcp` and value is > 0 then i read with `getDataBufTcp` everything works,
- since [here](https://github.com/adafruit/nina-fw/blob/master/arduino/libraries/WiFi/src/WiFiClient.cpp#L130) API is checking available anyway, i thought i can omit it in my app and stop reading from socket when `getDataBufTcp` return 0

I tracked down the problem to final issue. When i try to read from closed socket this is happening(endless loop):
```
[8]COMMAND: e02a010100ee0000
[8]COMMAND: e02f010100ee0000
[12]COMMAND: e045020001000002a00fee00
[8]COMMAND: e02e010100ee0000
[8]COMMAND: e02f010100ee0000
[12]COMMAND: e045020001000002a00fee00 // its the last command my app is sending, getDataBufTcp from closed socket
[4092]COMMAND: aa3cb8329d1034b6b0b3b297b53832b3868521b7b73a32b73a16a632b733ba341d101a199b9a9b068521b7b73732b1ba34b7b71d1035b2b2b81                                                           6b0b634bb32868525b2b2b816a0b634bb329d103a34b6b2b7baba1e991806852630b9ba16a6b7b234b334b2b21d1029bab71610191c10253ab6101918189c90181                                                           89d1a9c9d191b9023a6aa068522aa30b39d101130b0b2b1969a9c32b11a1918189b1b19b19c11068520b1b1b2b83a16a930b733b2b99d10313cba32b9868506857                                                           fec7ff0894822bc34b3000026a68015000000040003808900018000000080008000008d00028000000080000031008d80028000000080000035009400018000000                                                           080010000009880010000000e80000039009900010000000a00000047c3b4800200000000800000520000006800057e400000138800057e400000138820b237b13                                                           290283437ba37b9b437b81021a190142bb4b73237bbb994801918189c9d18191d191810189a9d19199d189980000001d0008001800000008000800050010002000                                                           00000800000a05001800200000000800000780000000000000003008180018000000080030000008d0002800000008000008f008d8002800000008000009300940                                                           00180000000800100000100800200000000800000970101000200000000800008ad0000000000000024000000008000002400000000ffec7ff6800620b237b132a                                                           fa1a68000fff7000720b237b13280324000000000ffed8042000604040404840604848608858505888a878606078a8c09898a89898c08860606060606088606060                                                           606060606060606060606060606060606060606060606060600868585868706880707080a0707070a0a070707070a0886060606060888860606060606088606060                                                           60606060606060606060606060606060606060606060606067fe0000884003c005001809100010880818880ffee800200057fe2009f80000082808080808080800                                                           0000000000001800081020283038404850580800082808080808080800000000000000080010182028303840485058800008200818102010283830402818619808                                                           00108818210891882a0a8b0899138c099030a48d0d8a111920aa960b1199a394168a18392c929f870f8b1b99a8b515941932249aa3222e151ba1b0be92af132f95                                                           9c261e9baf1f9a313ca5242da4ae26a727a52dae2eaf2faab333b434b535b636b737b1ba3abb3bbc3cbd3dbe3ebf3fb888001010081020201820283038383029a8                                                           0800108819098890220a8b0b89109829940c88a50d8a111e0a968f819923170b9414921a98ab1b99a7892830b51594183931ae1692249aa518bb222aa9b3a32f17                                                           959c261e9baf1f9a34a5242da4ae26a727a52dae2eaf2faab333b434b535b636b737b139ba3abb3bbc3cbd3dbe3ffed000601808001088188801f806ee25db46d3                                                           f44f7f02b427f8002c8a600d4ff057f1ed07f464ebf9a7f806fab041e8ee648f50dc0652d482a0c3c2934fd2f8a91869203a948712a33a9f827dd1a24866756c21                                                           ba64a27e6a1b8ea8a82dc823c9467b291d145c35f2412171a044981e6a0c83f052ef05141637110def36307fd20533ec6b18eb1da8634b9df2fcdfed28e55b5535                                                           b6db9e2ba98edd63dda35ad6b5ef2e71fecac7c7f802c74e4f503856d631d9b99663d85edb1b7d6e76ca6daaca9cea9dbaffa55b77f8027b3fd2b13e18a53b2058                                                           8eb241901a752e860baef45e321c5d66e9738f8b87bc6dfccf4de366d560c8d27855a878765e63068dadd5803b0259eff807a68e49d49ec25365f315c144e909e0                                                           0024a86ef149c4f29a9627527cd4883f22901f7f74898e124bfffe85ea5c0fe6fe0ae56e31c42a4cb9ff0566bb1fa69f82c02ef974b1aedfed77d31a9f31fbdff8                                                           01246075779954585e60821ae1ab4f81566ab0580e90e07dcdf03ff801177e6a7c4db14a3d5b0dd5338fb9269d545df4ff04e63a4c0c850727981f31f9533f9849                                                           43759c4fdc98927e5fc1271fee4c3b24966d200d549a1ef1b9c3d31f79f9c7b7f8033ef4275f563d62d7f8039cf1fa36773ff8018352c6becb71ecbde7ddee0d03                                                           40d007e684de2586fa389f59ba537a94d76665f4369da58c731afc6f508f6fa95d3b723d6ff8004fd3fa4b93c4e8fd4abc9c6b6eb58daf16c758c35b8bb6eedae7                                                           d55b5cd6edb3d6adaf73dcba8ea775ae7b5cd76e21dab498dc6367f805b8aad21f697d8e90d7400c7082607d3fde56e1ccca3cb9889f5ae1223d7f751ed0e30483                                                           dedb9d2f3a9c7379c8719b4b48305c499797b9dfe72da0f6bc6e690e6bc4b5c3821728e1b5c42d6e9196d6e25cdb09db410e681a98798dadffaefd1ff800c55b76                                                           521d3234f84a769926557fb5d7eb32a2d7b4d84b585c046e027d37439ceff80483ff4767e91e8d1ac76e51208dd0cc152f125439e3929c494829fffe8d8a9e0e84                                                           8856ea734707f1549847869f0562bd7c96110efb7d9622fa9a87b07bdbd8f0e1dd87f806f8a93227556590426ecb486fd4f6d803d865a7f803fe89fe52627dd3e6                                                           abd4ff49fbc6ac747a8d1e5fe107f29bf9ffbec4724120f239520361888a29678f881f8a4f3c7dca20e8df884ee2369ed08ad62f2036781127c16766f5ce9784df                                                           d68dae6ced25953dcdf86fdadadff803fec4d97d5fa6b3aa53d3acc8db90e6bbf46d236b5e21cc64                                                                                                             b4d8e3e8b5a3d2932d93fce3bfae99224648c251204871d9f4f147fa898faae8ec9727eb4f4accb9                                                                                                             cdc7b835a381602c7ba3fe336ffd6eb5afd1f28e4f48b1fb7690eb5a44cf017017d587682d2f63e7                                                                                                             f34901dfd92ba9fa93758fa33b02db3d42c2c7d64f3b6c6be9ff802aa94b2840409177a6ea90a15d                                                                                                             9a597638df73246df54910771e19bf7374f4dccb5bec629f4fbacf4c8b5db888efba091b9ed166d6                                                                                                             6f637f33d8b3faa5b93467bebc76b65ef6bdfbc120876d0f8702df4f637dca37f56c3c1616bdfbee                                                                                                             73b70a2bd5e01fa3ea7e655edfdf539c4678e2211e294a88006be29328c45934077f174722daeb06                                                                                                             db1c18c12493fc91bdfa0f73b6b06f5a1d069af2d9f6d63b7e398343c480e70ddfa56cff8021ddff                                                                                                             6eff80452e61f9d4db56275c633f478963b17a9547dc1a2dfd263ddff156377e3efdbfe0574f8ff5                                                                                                             ab0f26bdb88c75ae6375711b2b81ed6f8bff806b5b11972c71423237c5fa5feacfeeb18ca2722055                                                                                                             74feb78ba7531deb1263d3ae011041dc0691afd16ff826ff8003abfe16c472374155fa53dd6e27ab                                                                                                             6197d963dc [... cut ...]                                                                                00000000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                             00000000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                             000000000000000000000000000000000000000
```

ladyada, 2019-07-29:
could be! please add a check to the firmware that if a socket is closed, it will return an error :)

chegewara, 2019-07-29:
Like i said, im new to sockets, but as far as i know it could be possible to reuse the same socket, but if socket is closed because there is nothing more to read from it then we cant reuse it. 

Yes, now i am doing this:
- while
-- check if there is something to read from socket,
-- if not, check if we are still connected,
--- if not connected break
-- else read from socket

but this approach requires to use 2 additional SPI requests to check if socket is connected and there is something to read from socket.

Im not complaining, because i wouldnt make websocket app like this myself, just trying to figure out why i am having so many problems to use it with acceptable bandwidth and not loosing data.

ladyada, 2019-07-29:
it could be that this is the safest way to do it, so please use the extra calls for now if you dont want to make the changes

closed 2019-07-29 by chegewara