adafruit/nina-fw #9: esp32 wifi radio

chegewara, 2019-07-29:
With master branch i am having issue that esp32 wifi is going to sleep mode even if is used all the time:
```
I (220832) wifi: bcn_timout,ap_probe_send_start
I (223332) wifi: ap_probe_send over, resett wifi status to disassoc
I (223342) wifi: state: run -> init (c800)
I (223342) wifi: pm stop, total sleep time: 95335777 us / 217126450 us

I (223342) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (343352) event: station ip lost
```
`WiFi.noLowPowerMode();` is the solution.

ladyada, 2019-07-29:
is that a problem? maybe better if it goes to sleep in between uses and wakes up when CS pin is toggled?

chegewara, 2019-07-29:
Well, in my case it is problem, because it happens in middle of file downloading.

Im not saying its problem with nina-fw, i understand its problem with esp-idf.

My test case is to download file (about 40kB) from LAN server, and i am having many issues. To find root cause of issue i am downloading this file in loop (open socket, download, close socket). Sometimes file is not completely downloaded, always after many download esp32 just stop responding.

If you think this issue should be solved somewhere else, feel free to close it. Thanks.

ladyada, 2019-07-29:
hmmm! ok can you do a PR with a fix :) then we can test it!

chegewara, 2019-07-29:
Looks like some of my issues are related to this:
https://github.com/espressif/esp-idf/issues/3666

closed 2019-07-29 by chegewara