adafruit/nina-fw #19: WiFi.status() always WL_NO_MODULE

NightSkySK, 2020-03-15:
I'm struggling to make working communication between SAMD51 and ESP32. Basically my board is based on MetroM4 Airlift design. 
I wrote some more about my so far investigation if Adafruit Forum [Could not connect to ESP32 via SPI](https://forums.adafruit.com/viewtopic.php?f=8&t=163439)

To make a long story short:
I'm able to upload new firmware to ESP via esptool (I've tried 1.4.0, 1.5.0 and 1.6.0) 
[image]

I'm able to get a response from ESP32 with the firmware version through SPI correctly.

But calling WiFi.status() result is always WL_NO_MODULE on my board, once on my reference board Metro M4 Airlift with NINA FW 1.2.2 the communication with ESP is correct and I get list of AP, and can establish the connection.

I'm out of the idea how to check this further. 
@brentru could you check if the ESP chip which, I used is compatible with this FW Chip is ESP32D0WDQ6 (revision 1)? Or any other idea?

brentru, 2020-03-17:
@NightSkySK  Are you calling wifi.status after or before wifi.connect?

NightSkySK, 2020-03-17:
```
#include <Arduino.h>
#include <SPI.h>
#include <WiFiNINA.h>

void setup()
{
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial)
  {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.println("WiFi Scanning test");

  // Set up the pins!
  WiFi.setPins(SPIWIFI_SS, SPIWIFI_ACK, ESP32_RESETN, ESP32_GPIO0, &SPIWIFI);

  // check for the WiFi module:
  while (WiFi.status() == WL_NO_MODULE)
  {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    delay(1000);
  }
  String fv = WiFi.firmwareVersion();
  Serial.println(fv);
  if (fv < "1.0.0")
  {
    Serial.println("Please upgrade the firmware");
    while (1)
      delay(10);
  }
  Serial.println("Firmware OK");
}

void loop()
{
}
```
The same code is working on the original Metro M4 Airlift. The code (shorten) is from examples attached to the library WiFiNINA

NightSkySK, 2020-03-18:
After deeper investigation it seems that issue is hardware related.

closed 2020-03-18 by NightSkySK