adafruit/nina-fw #26: v1.7.0 not working with HCI

jerryneedell, 2020-08-31:
I was testing the the Circuitpython _bleio using the ESP32 as per https://github.com/adafruit/circuitpython/pull/3310

on a metro_m4_airlift_lite

It works with the NINA HCI BT firmware attached to the PR
I tried building it myself and create NINA_W102-1.7.0.bin
but when I try it, it fails:

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.3-48-gf5a5fc4c8 on 2020-08-31; Adafruit Metro M4 Airlift Lite with samd51j19
>>> from adafruit_ble import BLERadio
>>> ble = BLERadio()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ble/__init__.py", line 163, in __init__
  File "adafruit_ble/hci.py", line 109, in create_adapter
  File "adafruit_ble/hci.py", line 91, in create_adapter_esp32_hci
_bleio.BluetoothError: Timeout waiting for HCI response
>>> 
```

Note, the WiFi still works
```

Adafruit CircuitPython 6.0.0-alpha.3-48-gf5a5fc4c8 on 2020-08-31; Adafruit Metro M4 Airlift Lite with samd51j19
>>> 
>>> import esp32spi_simpletest
ESP32 SPI webclient test
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.7.0\x00')
MAC addr: ['0xdd', '0xa9', '0xe', '0x33', '0x4f', '0xc4']
        Needell Airport         RSSI: -58
        Needell Airport         RSSI: -74
        WYZE_FBAB62DE71811B38           RSSI: -88
Connecting to AP...
Connected to Needell Airport    RSSI: -55
My IP address is 10.0.0.70
IP lookup adafruit.com: 104.20.39.240
Ping google.com: 30 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)

----------------------------------------

Fetching json from http://api.coindesk.com/v1/bpi/currentprice/USD.json
----------------------------------------
{'time': {'updated': 'Aug 31, 2020 12:23:00 UTC', 'updatedISO': '2020-08-31T12:23:00+00:00', 'updateduk': 'Aug 31, 2020 at 13:23 BST'}, 'disclaimer': 'This data was produced from the CoinDesk Bitcoin Price Index (USD). Non-USD currency data converted using hourly conversion rate from openexchangerates.org', 'bpi': {'USD': {'code': 'USD', 'description': 'United States Dollar', 'rate_float': 11700.3, 'rate': '11,700.2707'}}}
----------------------------------------
Done!

brentru, 2020-09-07:
@jerryneedell  I am getting the same failure as you. WiFi works.

```
Adafruit CircuitPython 6.0.0-alpha.3-95-ga9779b99f on 2020-09-07; Adafruit Metro M4 Airlift Lite with samd51j19

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 5, in <module>
  File "/lib/adafruit_airlift/esp32.py", line 170, in start_bluetooth
_bleio.BluetoothError: Timeout waiting for HCI response
```

jerryneedell, 2020-10-26:
Fixed with 1.7.1

closed 2020-10-26 by jerryneedell