adafruit/nina-fw #28: WPA2 Enterprise Support

rayquazafan, 2020-11-03:
Will WPA2 Enterprise support be included anytime soon??

docmollo, 2020-11-04:
Hi there,

Sorry for the delay - my day got away from me.

While I haven't poked at it in a long while, ass the person who originally added the WPA2 Enterprise support to the Adafruit port of the NINA firmware and to CircuitPython, it at least was supported. The code does appear to still be there in the NINA port and CP. Have you tried the [code example](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/examples/esp32spi_wpa2ent_simpletest.py)?

Let us know how it goes.

rayquazafan, 2020-11-06:
Hello,

I was trying to get WPA2 Enterprise working on the Arduino side of things.
I tried to run the ConnectWPA2Enterprise example after adding the Enterprise related functions (from the arduino branch) to the adafruit fork of the WiFiNINA library, but still no success.
So I assumed something must be done in the firmware level as well.

docmollo, 2020-11-06:
Hi there,

As far as I know, Adafruit doesn't test or officially support their fork of the NINA fw with Arduino. If you're goal is to use Arduino to talk to an ESP32 running NINA fw, I think the Arduino version of NINA is going to be your best bet. I believe they added WPA2 Enterprise support several months ago.

-dustin