adafruit/nina-fw #30: Debug instructions needed

ricardoquesada, 2020-11-11:
Hi,

I've reading the different docs, source code, etc, and I couldn't find any doc that mentions how to debug the nina-fw.

For reference:
- I have a Matrix Portal M4
- I need to make some changes to the nina-fw (I want to enable bluetooth-classic)
- But I don't know how to debug it... I'm not asking for gdb... Just a pure "printf" is enough, but it seems that UART is not exposed?

How do you Adafruit developer debug nina-fw in Matrix Portal M4?
thanks!

If this is not the right place, where should I ask/post these kind of questions?, thanks!!

ladyada, 2020-11-11:
there is a debug output on the *esp32* RX / TX pins - you could use debug printf there
https://github.com/adafruit/nina-fw/blob/master/main/sketch.ino.cpp#L96
note that for BLE, at least, we sorta just jump into the HCI manager that does BLE and it doesnt appear anyone has gotten BT classic going yet
https://github.com/arduino/nina-fw/issues/35
it would be super cool!

ricardoquesada, 2020-11-11:
Ah...  GPIO1/GPIO3 (TX/RX) are enabled in runtime. That makes sense.

One more: How can I read from these pins?
Should I attach my serial console to the TX/RX pins that are exposed in the Matrix Portal M4 board?

(Just attaching a console to `/dev/ttyACM0` doesn't seem to work).

--

I already have BTClassic working in my own firmware. What I'm doing is adding Nina compatibility to my firmware: it only needs to understand some of the SPI commands, plus some extra ones related to gamepad:
https://gitlab.com/ricardoquesada/bluepad32/blob/master/src/main/uni_platform_airlift.c

I want to convert the Matrix Portal M4 + LED Matrix into a game console by adding gamepad support.
(Any Airlift based product will be supported).

Thanks!

dglaude, 2020-11-11:
The TX and RX exposed are those of the M4, not of the ESP32. Like the passthrough solution, you could have code running on the M4 that copy that over to those pin (read from one UART and write on the other). 

From https://learn.adafruit.com/assets/95095 : ESP_RXD (29) and ESP_TXD (30) are connected to PA12/I12/I2C/SERCOM2.0+4.1 and PA13/I13/I2C/SERCOM2.1+4.0. (I love how openhardware and having the schematics car help).

ricardoquesada, 2020-11-11:
many thanks!

ricardoquesada, 2020-11-21:
Many thanks, I was able to have it running. Source code of the new firmware is here:
https://gitlab.com/ricardoquesada/bluepad32/

I'll publish docs, examples and the modified `adafrui_esp32spi` later.

ladyada, 2020-11-21:
thats awesome! congrats. please update this thread when you publish and we'll blog it up

closed 2020-11-21 by ladyada

ricardoquesada, 2020-11-22:
Here are the docs:
https://gitlab.com/ricardoquesada/bluepad32/-/blob/master/docs/plat_airlift.md

Here is a patch for adafruit_esp32spi:
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/118

for reference, video showing it working:
https://www.youtube.com/watch?v=h4na4A5wnJ8

This is the list of supported gamepads: 
https://gitlab.com/ricardoquesada/bluepad32/-/blob/master/docs/supported_gamepads.md
(DS5 coming soon™)

There are still a few features in my TODO list like: possibility to delete bluetooth keys, gamepad LEDs/color & rumble,...
I'll be adding them in the coming weeks.

Please, let me know if you have any question / doubt / etc.
If anyone try it, I'd love to get feedback (what would you add, change, etc.).

Thanks!

ricardoquesada, 2020-11-23:
mini update: I've just added DualSense (PS5) gamepad support: https://www.youtube.com/watch?v=dbEbiJZd4n8

ladyada, 2020-11-23:
cool, please see if there's some way we can merge it into nina :)

ricardoquesada, 2020-11-23:
ha! for sure, that would nice! Although I see some possible issue:

The Bluepad32 project is a medium-sizde project (~8000 LoC) without including BTstack code (external dependency). Not sure whether someone from the Adafruit team would like to review that many LoC.

As of today, Bluepad32 only supports two platforms: Unijoyticle (gamepad for retro-computers) and AirLift. But my plan is keep adding support for other platforms... a goal different from Nina's.

Also, if we copy only the relevant parts of Bluepad32 into Nina (kind of a fork), then the future updates that I'll make to Bluepad32 (let's say adding a new gamepad, fixing bugs, etc.) won't be automatically included in Nina.

It might be easier to merge Nina into Bluepad32 (merge 1000 Loc from `CommandHandler.cpp` + some Arduino libs I guess). But I rather not have the Arduino-for-ESP32 code in Bluepad32 just to have it as simple as possible.

An alternative would be to create a bluepad32-for-airlift static lib + headers to be used at compile/link time in Nina.

What do you all have in mind? Thanks.

ladyada, 2020-11-23:
you could create a new esp32 binary for folks to load into their airlift board - we need to keep this repo so we can update from upstream :)

ricardoquesada, 2020-11-24:
done: https://github.com/ricardoquesada/bluepad32/releases/tag/release_v2.0.0-beta0  :)

The .zip includes a README with the steps to flash it.

bettse, 2021-11-07:
For anyone coming along later, using platformIO, the ESP32's uart has been exposed in the variant.cpp (which I believe is just being imported from the arduino version) as `SerialESP32`.  I found success outputting the ESP32 uart using

```c
while (SerialESP32.available()) {                                                                                                                                                                                                                                
  Serial.write(SerialESP32.read());                                                                                                                                                                                                                              
}
```