adafruit/nina-fw #39: Arduino Nano RP2040 Connect compatibility

matemaciek, 2021-10-12:
Is this fork compatible with Arduino Nano RP2040 Connect? I see that original repo has seperared builds for this board, so I guess I'd need a dedicated build here to?

ladyada, 2021-10-12:
nope! we use a different pinout - you should stick to arduino's firmware, its *very specialized* on the rp2040

matemaciek, 2021-10-12:
Thanks for a quick reply! Is it possoble to use BLE in CircuitPython via original arduino's firmware?

ladyada, 2021-10-12:
no idea :/ you could try it!

matemaciek, 2021-10-12:
It looks like it doesn't, or I'm doing something wrong )-:
```python
import board
from adafruit_airlift.esp32 import ESP32
esp32 = ESP32(tx=board.TX, rx=board.RX)
adapter = esp32.start_bluetooth()
```
results in:
```
Traceback (most recent call last):
  File "code.py", line 4, in <module>
  File "adafruit_airlift/esp32.py", line 185, in start_bluetooth
_bleio.BluetoothError: Timeout waiting for HCI response
```