adafruit/nina-fw #47: Installing NINA_W102-1.7.4 stops WiFi

Givy, 2022-07-24:
Hi,
Installation of "NINA_W102-1.7.4" on Airlift Breakout stops the WiFi operation on this module.  No WiFi Sketch works on Metro Express M4 with Airlift Breakout attached after this update.  See below update result.  Is there a Fix/re-course to this update?

[Airlift-NinAUpdate-Results.txt](https://github.com/adafruit/nina-fw/files/9176427/Airlift-NinAUpdate-Results.txt)

Many thanks.

Givy, 2022-07-30:
I bought two of these.  How do we put the Alitlift back into operation for WiFi and Blue tooth?  Is there a safe firmware we can install, and where do I get that?

Thanks.

ladyada, 2022-08-01:
older versions are here https://github.com/adafruit/nina-fw/releases

Givy, 2022-08-03:
> In general, v1.74 has worked fine, so this might be peculiar to your particular setup.
>
>   1.  Are you being guided by the instructions here? https://learn.adafruit.com/adafruit-airlift-breakout/upgrade-external-esp32-airlift-firmware<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Flearn.adafruit.com%2Fadafruit-airlift-breakout%2Fupgrade-external-esp32-airlift-firmware&data=05%7C01%7C%7C152e483c07974e4ef5b308da73d27598%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637949644995561318%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=GY3vFUThwJ8AhGgn43ILB2civLo%2FEXBcZPbYq3rcBNE%3D&reserved=0>
>   2.  Could you try an esptool.py ... erase_flash ... and then reload 1.74?
>   3.  Why did you choose a baudrate of 2400 instead of the suggested 115200?
>   4.  If you replace the firmware with an older version from https://github.com/adafruit/nina-fw/releases<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fadafruit%2Fnina-fw%2Freleases&data=05%7C01%7C%7C152e483c07974e4ef5b308da73d27598%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637949644995561318%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=adj80gC8CCRew6i9bKvfyBQWWVQgcCC1K%2FJZdPqvHcI%3D&reserved=0>, does it work?
>   5.  What test program are you using to test the firmware's operation? Could you try https://learn.adafruit.com/adafruit-airlift-breakout/arduino#first-test-3024934<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Flearn.adafruit.com%2Fadafruit-airlift-breakout%2Farduino%23first-test-3024934&data=05%7C01%7C%7C152e483c07974e4ef5b308 [... cut ...] 598%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637949644995561318%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=DmbpuvokwbeiBhQd8SBWH1CHfRCCokjkQwASxmTv2WQ%3D&reserved=0> ?
>
> Thanks for your patience while we try to figure this out.

Hi Dan,

I tried a few times and each time I got the same messages from both Airlift Breakouts, see below.  I am not sure how to turn the Airlift ON and it seems the Serial Tx & Rc pins don't respond as you see below.  So, if there is a way, please let me know and I'll try it.

\AppData\Roaming\Python\Python39\site-packages> python.exe -u -m esptool --port COM11 --before no_reset --baud 2400 write_flash 0 S:\Downloads\NINA_W102-1.7.3.bin
esptool.py v4.1
Serial port COM11
WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

______________________________________________________________________
______________________________________________________________________

\AppData\Roaming\Python\Python39\site-packages> python.exe esptool --port COM11 --chip auto --baud 9600 erase_flash
esptool.py v4.1
Serial port COM11
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.

I will look more into this and see if there is anything else on below link that I can try.  Thanks.

Troubleshooting - ESP32 - — esptool.py latest documentation (espressif.com)<https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html>

Regards,

Givi

Troubleshooting - ESP32 - — esptool.py latest documentation<https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html>
Troubleshooting . Flashing problems can be fiddly to troubleshoot. Try the suggestions here if you’re having problems: Bootloader Won’t Respond
docs.espressif.com

Givy, 2022-08-05:
Hi Dan,

After several trials, replacing wires, I connected the CS pin to "+" and the GPIO0 pin to "-" on the Airlift.  Then ran below command and others, but I got the same response each time:

C:\AppData\Roaming\Python\Python39\site-packages> python.exe -u -m esptool --port COM11 --chip ESP32 --before no_reset --baud 9600 write_flash 0 S:\Downloads\NINA_W102-1.7.3.bin
esptool.py v4.1
Serial port COM11
WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting............
Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\__main__.py", line 11, in <module>
    esptool._main()
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\__init__.py", line 995, in _main
    main()
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\__init__.py", line 637, in main
    esp = esp or get_default_connected_device(
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\__init__.py", line 886, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\loader.py", line 603, in connect
    last_error = self._connect_attempt(
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\loader.py", line 544, in _connect_attempt
    self.sync()
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\loader.py", line 407, in sync
    val, _ = self.command(
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\loader.py", line 337, in command
    self.write(pkt)
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\esptool\loader.py", line 289, in write
    self._port.write(buf)
  File "C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

Any ideas, please let me know.

Regards,

Givi

Givy, 2022-09-08:
Hi Dan,
Sorry about the delay.  I looked into the "espressif" website for a way to completely reset the Airlift, but I was not successful.  The bootstrap is very confusing and not much practical information.  I looked in Adafruit web page and saw a new way (to me) of updattinig the firmware by using the "Google Chrome" and CIRCUITPY, please see attached files and test results.
https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html

Regards,
Givi
[AirliftConnections.pdf](https://github.com/adafruit/nina-fw/files/9528590/AirliftConnections.pdf)
[AirLift-Firmware-Google-Chrome.txt](https://github.com/adafruit/nina-fw/files/9528591/AirLift-Firmware-Google-Chrome.txt)
[image]

Givy, 2022-09-08:
Dan,
I also tried with the "ESP_DOWNLOAD_TOOL_V2.4", but it just sat there at "connecting...", please see attached.
[image]

Givy, 2022-09-11:
Hi Dan,
I used the ESP32-WROOM-32 Pin layout from ESP32-WROOM-32 Datasheet and connected the pins to a FTDI (USB) adapter (similar to  https://electronics.stackexchange.com/questions/448187/esp32-with-ftdi-programmer) :

FTDI PIN ---------------> ESP32 PIN
CTS -----------------> GPIO 15
RTS -----------------> GPIO 13 
GND --------------->GPIO 0
RX -----------------> TXD0
TX -----------------> RXD0
DTR -----------------> EN
VCC -----------------> Vin (5V)
GND -----------------> GND

Then used ESP32 DOWNLOADTOOL and Erased and uploaded the same FW again and it looked good and no issues.  See attached outputs.  Tested it with Python esptool as well and it seems it has taken and should be working.  I may have used the start of the memory (see the attached pic) and it might be wrong address(0X1000).  Let me know.

However, when I tested the Airlift with the Arduino Sketches it was not workiing at all.
[image]
_______________________________________________________________

\AppData\Roaming\Python\Python39\site-packages> python.exe esptool --chip esp32 --port COM4 --baud 115200 read_flash_status
esptool.py v4.1
Serial port COM4
Connecting...
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:9f:6d:d3:0f:74
Stub is already running. No upload is necessary.
Status value: 0x0000
Hard resetting via RTS pin...

Regards,
Givi

Givy, 2022-10-11:
Hi Dan,

Thanks for getting back with me on this.

I-A.  Yes, I have followed that link and its instructions.  Before applying the Firmware, I used examples provided for Arduino from Adafruit WiFiNINA.  I had a couple of cases opened with the Forum as well on this and other issues with the Airlift Breakout module.

2-A.  Yes, I will try to use "esptool.py...erase_flash", etc and may reload 1.74 again.  Not very happy with what it did.

3-A.  I chose lower Baud Rate because I read there will be less electrical noise generated in the connections on the breadboard and the jumper wires.  the Baud rate would not affect anything here but the speed of this process.

4-A.  I have not had a chance to do so, but I most likely would rather go back to a well-known working version.  I will follow your link and will get back with you on that.

5-A.  As soon as I get things installed and in a stable condition, I will follow your link to run the test/s and will let you know.

Thanks again.

Regards

Givi

Givy, 2022-10-30:
Hi Dan,

Still it is not functioning.
===================Latest erase and Update by ESP tool in GOOGLE========================

esptool.js v0.1-dev
Serial port WebSerial VendorID 0x403 ProductID 0x6001
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:9f:6d:d3:0f:74
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.352s
Compressed 1159172 bytes to 633594...
Writing at 0x0... (2%)
Writing at 0x4000... (5%)
Writing at 0x8000... (7%)
Writing at 0xc000... (10%)
Writing at 0x10000... (12%)
Writing at 0x14000... (15%)
Writing at 0x18000... (17%)
Writing at 0x1c000... (20%)
Writing at 0x20000... (23%)
Writing at 0x24000... (25%)
Writing at 0x28000... (28%)
Writing at 0x2c000... (30%)
Writing at 0x30000... (33%)
Writing at 0x34000... (35%)
Writing at 0x38000... (38%)
Writing at 0x3c000... (41%)
Writing at 0x40000... (43%)
Writing at 0x44000... (46%)
Writing at 0x48000... (48%)
Writing at 0x4c000... (51%)
Writing at 0x50000... (53%)
Writing at 0x54000... (56%)
Writing at 0x58000... (58%)
Writing at 0x5c000... (61%)
Writing at 0x60000... (64%)
Writing at 0x64000... (66%)
Writing at 0x68000... (69%)
Writing at 0x6c000... (71%)
Writing at 0x70000... (74%)
Writing at 0x74000... (76%)
Writing at 0x78000... (79%)
Writing at 0x7c000... (82%)
Writing at 0x80000... (84%)
Writing at 0x84000... (87%)
Writing at 0x88000... (89%)
Writing at 0x8c000... (92%)
Writing at 0x90000... (94%)
Writing at 0x94000... (97%)
Writing at 0x98000... (100%)
Wrote 1159172 bytes (633594 compressed) at 0x0 in 12.042 seconds.
Hash of data verified.
Leaving...

Givy, 2022-10-30:
Hi Dan,
Below is a try by Python esptool to erase and re-load the FW (NINA_W102-1.7.4.bin).  its the first time I see it comes up with "A fatal error occurred: Invalid firmware image magic=0xff"::
_______________________________________________________________

\AppData\Roaming\Python\Python39\site-packages> python.exe esptool --chip esp32 --port COM3 --baud 115200 --after hard_reset erase_flash
esptool.py v4.1
Serial port COM3
Connecting...
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:9f:6d:d3:0f:74
Stub is already running. No upload is necessary.
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.3s
Hard resetting via RTS pin...
PS C:\Users\Engineer\AppData\Roaming\Python\Python39\site-packages> python.exe esptool --chip esp32 --port COM3 --baud 115200 --after hard_reset load_ram S:\Downloads\NINA_W102-1.7.4.bin
esptool.py v4.1
Serial port COM3
Connecting...
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:9f:6d:d3:0f:74
Stub is already running. No upload is necessary.

A fatal error occurred: Invalid firmware image magic=0xff

Thanks AGAIN
Givi

dhalbert, 2025-02-19:
Closing as old. Sorry I was not able to solve this at the time.

closed 2025-02-19 by dhalbert

Givy, 2025-02-21:
Thank you Dan for looking into this 🙏

Sent from my Verizon, Samsung Galaxy smartphone

Closed #47<https://github.com/adafruit/nina-fw/issues/47> as completed.