adafruit/nina-fw #49: ESP32C3 GTS SSL Root certs not trusted (ie google.com not connecting)

Bruno-DaSilva, 2022-08-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-131-gf9b9f5568-dirty on 2022-08-13; Wemos Lolin C3 Mini with ESP32-C3FH4
```

### Code/REPL

```python
import wifi
import socketpool
import ssl

wifi.radio.connect('ssid', 'password')
pool = socketpool.SocketPool(wifi.radio)

import adafruit_ntp
import rtc
ntp = adafruit_ntp.NTP(pool, tz_offset=0)
rtc.RTC().datetime = ntp.datetime

def connect_https_socket(host, port, pool):
    addr_info = pool.getaddrinfo(host, port, 0, pool.SOCK_STREAM)[0]
    connect_host = addr_info[-1][0]
    sock = pool.socket(addr_info[0], addr_info[1])
    ssl_context = ssl.create_default_context()
    sock = ssl_context.wrap_socket(sock, server_hostname=host)
    connect_host = host
    try:
        sock.connect((connect_host, port))
    finally:   
        sock.close()

connect_https_socket('good.gsr4demo.pki.goog', 443, pool) # success
connect_https_socket('good.r1demo.pki.goog', 443, pool) # fails
# Traceback (most recent call last):
#   File "<stdin>", line 1, in <module>
#   File "<stdin>", line 9, in connect_https_socket
# OSError: Failed SSL handshake
connect_https_socket('good.r2demo.pki.goog', 443, pool) # fails
# Traceback (most recent call last):
#   File "<stdin>", line 1, in <module>
#   File "<stdin>", line 9, in connect_https_socket
# OSError: Failed SSL handshake
connect_https_socket('good.r3demo.pki.goog', 443, pool) # success
connect_https_socket('good.r4demo.pki.goog', 443, pool) # success

# Other examples
connect_https_socket('google.com', 443, pool) # failure
connect_https_socket('play.google.com', 443, pool) # failure
connect_https_socket('io.adafruit.com', 443, pool) # success
connect_https_socket('www.adafruit.com', 443, pool) # success
connect_https_socket('github.com', 443, pool) # success
```

### Behavior

Running this code function `connect_https_socket` will either produce 
1. no error if successful, or 
3. a failed handshake error

Depending on the destination. These are obviously very valid certificates (literally for *.google.com). 

### Description

Looks like its likely some root certificates are not in the firmware generated by the build process for the esp32. I checked and looks like GTS Root R1 was specifically added on this commit to the mozille certdata.txt on `5 Dec 2021` https://hg.mozilla.org/releases/mozilla-beta/annotate/d7c8bc02bda4c5cbeacf2b165e24db7e0ba345c2/security/nss/lib/ckfw/builtins/certdata.txt#l22997

Certs are here: https://pki.goog/repository/

I encountered this on a personal domain proxied through cloudflare that had an ssl cert with the GTS Root R1 root cert.

### Additional information

Not sure if related to this fix implemented here (I suspect not): https://github.com/adafruit/circuitpython/issues/3424

jepler, 2022-08-24:
I've transferred this issue to nina-fw because that's where the update needs to take place. After that, we'll need to update the nina-fw submodule in circuitpython as well.

Bruno-DaSilva, 2022-08-24:
Great thanks @jepler!