adafruit/nina-fw #58: Fails with https://api.intra.io/api/v1/healthz

dhalbert, 2023-09-21:
https://api.intra.io/api/v1/healthz, which has a Let's Encrypt cert chain, causes failure. It is Cloudflare based.

The cert chain as reported by Firefox is:
*.intra.io server cert <- Let's Encrypt E1 <- ISRG Root X2 (using elliptic keys)

This contrasts with the working Let's Encrypt chain:
End-entity certificate <- Let's Encrypt R3 intermediate cert <- ISRG Root X1 (special cross-signed version) <- DST Root CA X3
(see https://github.com/adafruit/certificates/pull/1 for background)

The URL above _does_ work on native Espressif wifi, like ESP32-S2.

Turning on debug tracing shows what might be a crash:
```
Fetching and parsing json from https://api.intra.io/api/v1/healthz
*** Get host by name
*** Get socket
Allocated socket #0
*** Socket connect mode 2
*** Open socket to api.intra.io 443 2
*** Closing socket #0
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
  File "adafruit_requests.py", line 748, in connect
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 77, in connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 819, in socket_connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 715, in socket_open
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 341, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 297, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 206, in _wait_for_ready
TimeoutError: ESP32 not responding
```

The cert chain that doesn't work uses elliptic keys, but those encryption methods seem to be turned on in the `sdkconfig` here.

Thanks to @crhuber for reporting this. See https://github.com/adafruit/nina-fw/issues/55#issuecomment-1727059504 and following.

@tyeth you mentioned two URL's you are using, one with each chain. If you have time, could you verify that one works and one fails, in line with the above?

dhalbert, 2023-09-24:
Fails with #57 (labeled as 1.7.7 tentatively), and also 1.7.4 and 1.7.5 NINA-FW builds, so it's not a new cert problem.

crhuber, 2023-09-25:
@dhalbert the only thing i can suspect about this is that it is some CloudFlare related thing like the size of the headers, http protocol config

dhalbert, 2023-09-25:
I found out Let's Encrypt has set up test websites that use both of its chains. See https://letsencrypt.org/certificates/, and look under "Root Certificates" for "We've set up websites..." to find them.

https://valid-isrgrootx1.letsencrypt.org/
chain: server <- Let's Encrypt R3 intermediate <- ISRG Root X1 (and to CA Root CA X3 if needed)
uses RSA keys

https://valid-isrgrootx2.letsencrypt.org/
chain: server <- Let's Encrypt E1 intermediate <- ISRG Root X2
uses ECDSA keys

The first server works fine, as expected. The second fails in the same way that api.intra.io fails. These servers don't seem to use CloudFlare -- they are straightforwardly configured.

So it appears the problem has to do with this particular cert chain, maybe because it is ECDSA.

dhalbert, 2023-09-25:
I tried the test servers using an Arduino Nano 33 IOT, with its https://github.com/arduino/nina-fw firmware updated to 1.5.0, the latest version. As with our test, https://valid-isrgrootx1.letsencrypt.org/ worked, but https://valid-isrgrootx2.letsencrypt.org/ did not. I was using the `WiFiSSLClient` example program. When trying the `valid-isrgrootx2.letsencrypt.org` server, the example program simply printed:
```
Attempting to connect to SSID: PBE2
Connected to WiFi
SSID: [redacted]
IP Address: [redacted]
signal strength (RSSI):-46 dBm

Starting connection to server...

disconnecting from server.
```
By comparison, with `valid-isrgrootx2.letsencrypt.org`, the HTML of the home page was printed out.

So upgrading Adafruit's NINA-FW to Arduino's version is not going to fix this problem.

tyeth, 2023-09-26:
@dhalbert I did plan to have a quick look, but got distracted while looking at the linked issues (esp-idf or esp-arduino one) and the minimum test case they asked for, I found both the sites worked with just the esp mbedtls CMN(common) certs included using the x509 example, I haven't tried out our bundle specifically in esp-idf.

dhalbert, 2023-09-26:
@tyeth Since https://valid-isrgrootx2.letsencrypt.org/ works with regular ESP32 and ESP32-xx (but does not work with either adafruit or arduino NINA-FW) we might look to see if there are some differences in the settings between the `sdkconfig` here and the one in CircuitPython. But the one here does turn on ECSDA in various forms, and I don't see a suspicious `=n`.

crhuber, 2023-12-04:
@dhalbert any updates on this one?

dhalbert, 2023-12-04:
@crhuber After further testing, I found that arduino NINA-FW does work, when the proper root certificates are included. We think we will rebase our fork of NINA-FW against the latest arduino NINA-FW and add back the features we added. But this work has not been started yet.

We forked a long time ago and there have been many changes to arduino NINA-FW since then. Our code changes are relatively minor.

terop, 2025-01-07:
@dhalbert any update on this? I have the same problem as in issue #66.
It looks like the upstream sync effort in PR #62 isn't going anywhere, not at least very quickly.

dhalbert, 2025-01-07:
I have asked about #62 but haven't gotten a reply yet.

Whisper380, 2025-01-17:
Any update on this issue?
 I’ve post issue 66 but still no solution. Is there any sign it will be resolved anywere this or the following Month? Or Will it take longer? 
I can not continue my project if I can not Connect to my websites. It really feels like a waste to buy another system because of this bug.

rad600, 2025-01-23:
Recently have seen a problem attempting to connect to globalnews.ca
I believe this issue is probably related to #58. Platform here is Adafruit esp32 wifi shield with 1.7.7 fw over a metro m4.
Global news cert reports it is ISRG Root X1 but as a X9.62 ECDSA Signature with SHA-384.
Thoughts? Any updates on #58?
Thanks for your time.

dhalbert, 2025-01-23:
I am debugging #62 as we speak 😄 .

rad600, 2025-02-12:
Referencing previous issue reported, downloaded/flashed latest release 2.0.0 to device and I'm happy to report that this has corrected the issue. Thanks to all for your your extensive time resolving this.

rad600, 2025-02-14:
Further testing. While 2.0.0 has resolved the previous issue with ECDSA, I find that it has broken a connection that worked previously with 1.7.7 and previous versions. The url in question is weather.gc.ca using a Entrust Certification Authority - L1K cert..
Reverting the firmware back to 1.7.7 on the esp-32 wifi shield corrects the issue. Possibly this cert was left out of the 2.0.0 build?
Again, thanks for your time on this.

dhalbert, 2025-02-14:
@rad600 Hi - you posted but apparently then deleted a comment early today about:

> The url in question is weather.gc.ca using a Entrust Certification Authority - L1K cert.

I removed Entrust certificates because there were discredited. See https://github.com/adafruit/certificates/issues/7, which refers to https://blog.webnames.ca/google-and-firefox-to-stop-supporting-entrust-ssl-certificates-what-you-need-to-know/. Firefox and Chrome are trusting Entrust certificates issued before Oct 31, 2024, but not more recent ones. I discarded all of them. Is this a problem for you?

rad600, 2025-02-14:
Hello yes I had added the comment here but then removed it and added as a new issue #67 as I thought that it wasn't related.  I understand the reasoning behind the Entrust issue and thanks for sharing that info. However weather.gc.ca is using Entrust Certification Authority - L1K cert as of April 19, 2024 and times out on May 19 2025. I'm sure they will change their Cert. to one that is trusted once it expires. Yes, this is a issue for me as I need to access weather.gc.ca with, preferably the updated 2.0.0, as this latest fw release fixes other issues related to other sites. Options?
Thanks.

dhalbert, 2025-02-19:
Fixed by #62.

closed 2025-02-19 by dhalbert