adafruit/nina-fw #67: 2.0.0 and Entrust Certification Authority - L1K cert

rad600, 2025-02-14:
Further testing of 2.0.0. While 2.0.0 has resolved the previous issue with ECDSA, I find that it has broken a connection that worked previously with 1.7.7 and previous versions. The url in question is weather.gc.ca using a Entrust Certification Authority - L1K cert..
Reverting the firmware back to 1.7.7 on the esp-32 wifi shield corrects the issue. Possibly this cert was left out of the 2.0.0 build?
Again, thanks for your time on this.

dhalbert, 2025-02-14:
Copying from #58:

@rad600 Hi - you posted but apparently then deleted a comment early today about:

    The url in question is weather.gc.ca using a Entrust Certification Authority - L1K cert.

I removed Entrust certificates because they were discredited. See https://github.com/adafruit/certificates/issues/7, which refers to https://blog.webnames.ca/google-and-firefox-to-stop-supporting-entrust-ssl-certificates-what-you-need-to-know/. Firefox and Chrome are trusting Entrust certificates issued before Oct 31, 2024, but not more recent ones. I discarded all of them. Is this a problem for you?

---

from @rad600 [on Feb 14, 2025](https://github.com/adafruit/nina-fw/issues/58#issuecomment-2660323428)

Hello yes I had added the comment here but then removed it and added as a new issue https://github.com/adafruit/nina-fw/issues/67 as I thought that it wasn't related. I understand the reasoning behind the Entrust issue and thanks for sharing that info. However weather.gc.ca is using Entrust Certification Authority - L1K cert as of April 19, 2024 and times out on May 19 2025. I'm sure they will change their Cert. to one that is trusted once it expires. Yes, this is a issue for me as I need to access weather.gc.ca with, preferably the updated 2.0.0, as this latest fw release fixes other issues related to other sites. Options?
Thanks.

dhalbert, 2025-02-18:
The major browsers fudged things a bit by not discrediting server certificates derived from the Entrust root certificates and issued before Oct/Nov 2024. So ideally we would not do that either, but to restore the Entrust roots would cause regressions due to heap problems. So it's a dilemma. I am working on a few possibilities:

- Change how the root certs are parsed and loaded from all to once to one or a few at a time, to see if I can reduce the impact on the heap.
- Provide a way to load a custom set of root certs via a new API in the ESP32SPI library.
- Provide a procedure to replace the `roots.pem` root certificate data in the firmware. This would involve using esptool.py.
- Update to ESP-IDF 5.4, which might or might not provide an improvement in various ways.

rad600, 2025-02-22:
Thanks for this. I like the idea of loading custom certs and or replace the root certs data in fw. Ideally with a standalone or web based app. The latter similar in nature to the current web page available to upload fw.

dglaude, 2025-02-23:
I love the idea of updating the certificate without changing the firmware, or adding a custom certificate. It might offer quick work around and use of private PKI without recompiling.

rad600, 2025-05-06:
Any updates on cert updates without uploading firmware? Today weather.gc.ca changed their cert, staying with Entrust. Now the cert is Entrust OV TLS Issuing RSA CA 1. Of course this broke the connection here using FW 1.7.7 and 2.0.0. I see there is a AI generated update on Google's relationship with Entrust for what it is worth. "Current Status:
As of the current date, May 5, 2025, Chrome continues to trust Entrust certificates. While there was a period where Google was set to distrust new issuances, the situation has been resolved, and Entrust certificates are again considered trusted."
Maybe this is why gc.ca stayed with Entrust...

Thanks,

dhalbert, 2025-07-28:
3.1.0 just released, which fixes #71. I'll close this since we now supply the full Mozilla cert list. Please re-open or open a new issue if you still have cert trouble after upgrading.

closed 2025-07-28 by dhalbert

rad600, 2025-07-28:
Thanks Dan and everyone that contributed to the fix of both of issues, 67
and 71.