adafruit/nina-fw #71: Version 3.0.0 frequently fails HTTP requests

FoamyGuy, 2025-07-10:
I've been testing with this modified requests simpletest:
```

import os

import adafruit_connection_manager
import board
import busio
from adafruit_esp32spi import adafruit_esp32spi
from digitalio import DigitalInOut

import adafruit_requests

# Get WiFi details, ensure these are setup in settings.toml
ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

# If you are using a board with pre-defined ESP32 Pins:
esp32_cs = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
radio = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset, debug=True)
print(f"esp firmware version: {radio.firmware_version}")
print("Connecting to AP...")
while not radio.is_connected:
    try:
        radio.connect_AP(ssid, password)
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(radio.ap_info.ssid, "utf-8"), "\tRSSI:", radio.ap_info.rssi)

# Initialize a requests session
pool = adafruit_connection_manager.get_radio_socketpool(radio)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(radio)
requests = adafruit_requests.Session(pool, ssl_context)

JSON_GET_URL = "https://www.adafruit.com/api/products/6358"
# JSON_GET_URL = "https://httpbin.org/get"
# JSON_GET_URL = "https://fantasy.premierleague.com/api/bootstrap-static/"

print("Fetching JSON data from %s" % JSON_GET_URL)
with requests.get(JSON_GET_URL, timeout=15) as response:
    print("-" * 40)
    print("JSON Response: ", response.json())
    print("-" * 40)
```

I've tested with both ESP32 based Airlift Featherwing w/ RP2040, RP2350, and SAMD51, and ESP32-C6 based with Fruit-Jam, and all seem to behave the same way.

Using version 3.0.0 of nina-fw this code fails very frequently in one of a few ways.

SPI timeout:
```
Traceback (most recent call last):
  File "code.py", line 47, in <module>
  File "/lib/adafruit_requests.py", line 721, in get
  File "/lib/adafruit_requests.py", line 649, in request
  File "/lib/adafruit_connection_manager.py", line 328, in get_socket
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 60, in getaddrinfo
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 752, in get_host_by_name
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 422, in _send_command_get_response
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 389, in _wait_response_cmd
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 368, in _wait_spi_char
TimeoutError: Timed out waiting for SPI char

```

ETIMEDOUT exception raised and ESPSPI reporting that the socket has 0 bytes to read for a long stretch

```
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
.... repeated many times ...
ESPSocket: 0 bytes available
*** Closing socket #0
Traceback (most recent call last):
  File "/lib/adafruit_requests.py", line 665, in request
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 139, in recv
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 180, in recv_into
OSError: [Errno 116] ETIMEDOUT

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 47, in <module>
  File "/lib/adafruit_requests.py", line 721, in get
  File "/lib/adafruit_requests.py", line 681, in request
OutOfRetries: Repeated socket failures

```

If I downgrade the firmware on the Airlift Featherwing to 2.0.0 the same test script is able to run successfully over and over without the same problems.

FoamyGuy, 2025-07-10:
@hathach if you have a chance can you look into this? Let me know if there is any additional debugging info or anything else I could try and provide that would be of assistance.

I saw there are debug build .bin files in the release assets, would using those be helpful in figuring this issue out?

FoamyGuy, 2025-07-11:
I've tested and replicated the same issue now on local network with http instead of https as well. I used `python -m http.server` 

I loaded the debug firmware and captured this log file from the ESP32: [esp32_debugout.txt](https://github.com/user-attachments/files/21185138/esp32_debugout.txt)

On the circuitpython side these logs are produced by esp32spi debug:
```
... many similar lines...
Reading 64 bytes from ESP socket with status 4
ESPSocket: 674 bytes available
Reading 64 bytes from ESP socket with status 4
ESPSocket: 610 bytes available
Reading 64 bytes from ESP socket with status 0
Traceback (most recent call last):
  File "code.py", line 54, in <module>
  File "/lib/adafruit_requests.py", line 342, in json
  File "/lib/adafruit_requests.py", line 78, in readinto
  File "/lib/adafruit_requests.py", line 224, in _readinto
  File "/lib/adafruit_requests.py", line 143, in _recv_into
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 180, in recv_into
OSError: [Errno 116] ETIMEDOUT
```

Most of the time the server indicates that success response was returned `"GET /6358.json HTTP/1.1" 200 -`

hathach, 2025-07-14:
thank you for detail, I am able to reproduce the issue using your script with `metro rp2350 + C6`. I am doing more testing to see what could cause the issue.

```
ESPSocket: 0 bytes available                                                                                    
ESPSocket: 0 bytes available                                                                                    
*** Closing socket #0                                                                                           
*** Get host by name                                                                                            
*** Get socket                                                                                                  
Allocated socket #0                                                                                             
*** Socket connect mode 2                                                                                       
*** Open socket to www.adafruit.com 443 2                                                                       
*** Closing socket #0                                                                                           
Traceback (most recent call last):                                                                              
  File "code.py", line 42, in <module>                                                                          
  File "adafruit_requests.py", line 721, in get                                                                 
  File "adafruit_requests.py", line 649, in request                                                             
  File "adafruit_connection_manager.py", line 331, in get_socket                                                
  File "adafruit_connection_manager.py", line 248, in _get_connected_socket                                     
  File "adafruit_connection_manager.py", line 61, in connect                                                    
  File "adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 114, in connect                                
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 899, in socket_connect                                    
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 801, in socket_open                                       
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 422, in _send_command_get_response                        
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 392, in _wait_response_cmd                                
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 374, in _check_data                                       
BrokenPipeError: Expected 01 but got 00   
```

dhalbert, 2025-07-15:
The bundled vs custom root cert list should not really make any difference, as it's stored the same way internally. The way it's done in ESP-IDF 5.4.x is actually pretty efficient on RAM. I don't remember if there is enough checking for storage allocations or not in the current NINA impl.

hathach, 2025-07-16:
thank @dhalbert yeah it is something else, certifacte seems to work well. I enabled SSL debug log, but still not sure what cause it. I will try to test more with python simple server

```
I (7643) mbedtls: ssl_tls.c:4585 client state: MBEDTLS_SSL_FLUSH_BUFFERS
I (7650) mbedtls: ssl_tls12_client.c:3587 handshake: done
I (7655) mbedtls: ssl_msg.c:2354 => flush output
I (7660) mbedtls: ssl_msg.c:2363 <= flush output
I (7664) mbedtls: ssl_tls.c:4585 client state: MBEDTLS_SSL_HANDSHAKE_WRAPUP

I (9886) mbedtls: ssl_msg.c:5928 => read
I (9912) mbedtls: ssl_msg.c:4298 => read record
I (9916) mbedtls: ssl_msg.c:2156 => fetch input
I (9921) mbedtls: ssl_msg.c:2296 in_left: 0, nb_want: 5
I (9910) mbedtls: ssl_msg.c:5928 => read
I (9929) mbedtls: ssl_msg.c:4298 => read record
I (9934) mbedtls: ssl_msg.c:2156 => fetch input
I (9938) mbedtls: ssl_msg.c:2296 in_left: 0, nb_want: 5
I (9927) mbedtls: ssl_msg.c:2316 in_left: 0, nb_want: 5
I (9944) mbedtls: ssl_msg.c:2316 in_left: 0, nb_want: 5
I (9954) mbedtls: ssl_msg.c:2319 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (9961) mbedtls: ssl_msg.c:2341 <= fetch input
W (9965) mbedtls: ssl_msg.c:3920 unknown record type 0
W (9970) mbedtls: ssl_msg.c:4329 ssl_get_next_record() returned -29184 (-0x7200)
W (9978) mbedtls: ssl_msg.c:5988 mbedtls_ssl_read_record() returned -29184 (-0x7200)
[  9603][E][ssl_client.cpp:38] _handle_error(): [data_to_read():411]: (-29184) SSL - I (9950) mbedtls: ssl_msg.c:2319 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
An invalid SSL record was received
I (10000) mbedtls: ssl_msg.c:2341 <= fetch input
W (10008) mbedtls: ssl_msg.c:3920 unknown record type 0
W (10014) mbedtls: ssl_msg.c:4329 ssl_get_next_record() returned -29184 (-0x7200)
W (10021) mbedtls: ssl_msg.c:5988 mbedtls_ssl_read_record() returned -29184 (-0x7200)
[  9647][E][ssl_client.cpp:38] _handle_error(): [data_to_read():411]: (-29184) SSL - An invalid SSL record was received
[  9668][E][NetworkClientSecure.cpp:309] available(): Closing connection on failed available check
[  9678][E][NetworkClientSecure.cpp:309] available(): Closing connection on failed available check
I (10060) mbedtls: ssl_tls.c:5581 => free
I (10061) mbedtls: ssl_tls.c:5641 <= free
```

dhalbert, 2025-07-16:
I wonder if it is anything to do with the the ready/busy line not showing status properly. But that is just a wild guess.

hathach, 2025-07-16:
I test with local `python -m http.server` with cpy 10 times back-to-back and it seems to work well. Not sure which server my python use with though.

FoamyGuy, 2025-07-16:
I had the `python -m http.server` running on a Raspberry Pi 5 (connected to same network as MCU via WIFI) and hosting a downloaded copy of the JSON file returned by this adafruit product API call: https://www.adafruit.com/api/products/6358

hathach, 2025-07-17:
just test to get the json file, it seems to work well with python http.server form adafruit api. I guess the issue is more reproducible with ssl, will try to focus on that.  

```
Fetching JSON data from http://192.168.31.20:8000/pid6358.json                                              
----------------------------------------                                                                    
JSON Response:  {'product_price': '9.95', 'products_description': '<div class="row-fluid build-text"><p>You}
----------------------------------------                                                                    
0;🐍Done | 10.0.0-alpha.8-5-ge80c92cfaf-dirty                                                               
Code done running.                                              
```

dhalbert, 2025-07-17:
This is confusing to me, because it sounds like you both are testing fetching `pid6358.json` with `python -m http.server`, using the code in the original post, yet it's failing for @FoamyGuy and succeeding for @hathach. So what can be different, if anything, between these two setups? Maybe if you both describe the setup we can figure out the difference. 

@FoamyGuy did it fail fetching with that server for both ESP32-C6 and ESP32? 
@hathach on what platform were you running `python -m http.server`? Was it on an RPi or something else? @FoamyGuy could you trying running the server on, say, Windows, or a desktop Linux?

FoamyGuy, 2025-07-17:
I tested with `python -m http.server` running on my development machine ubuntu 24.04, and I am getting the same results as when I had that running on the Pi before. I tested both C6/FruitJam, and ESP32/RP2350 with same outcome on both. Still fetching the same static JSON file. 

```
ESPSocket: 738 bytes available
Reading 64 bytes from ESP socket with status 4
ESPSocket: 674 bytes available
Reading 64 bytes from ESP socket with status 4
ESPSocket: 610 bytes available
Reading 64 bytes from ESP socket with status 0
Traceback (most recent call last):
  File "code.py", line 55, in <module>
  File "/lib/adafruit_requests.py", line 342, in json
  File "/lib/adafruit_requests.py", line 78, in readinto
  File "/lib/adafruit_requests.py", line 224, in _readinto
  File "/lib/adafruit_requests.py", line 143, in _recv_into
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 180, in recv_into
OSError: [Errno 116] ETIMEDOUT
```

I noticed they both seem to consistently time out at the exact same spot, the last amount reported is `610 bytes available` It attempts the next read, gets status 0, then waits a few seconds and raises this timeout exception.  Both devices report exact same amount for me before failing. 

I'm not sure if it's relevant but my dev machine has both wifi and ethernet and I tested requesting from both different IPs and both seem to have identical behavior, fail at the same place and everything else the same.

I think I can try on windows as well, I'm also going to try experimenting with a dynamic response and see if I can correlate any specific response size to the failure.

dhalbert, 2025-07-17:
I tested with 3.0.0 on:
- AirLift FeatherWing. All jumper pads on the bottom of the board are soldered closed.
- Feather RP2040
- Test program as in the first, but with different values `DEBUG=True` and `DEBUG=False` on the line:
```py
radio = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset, debug=DEBUG)
```
**Crucially, it works fine with `debug=False`, and I get `ETIMEDOUT` with `debug=True`**.

So there is something happening when `debug=True` that is causing a problem.

FoamyGuy, 2025-07-17:
I do get consistently successfully responses with both `http://wifitest.adafruit.com/testwifi/sample.json` and `http://wifitest.adafruit.com/testwifi/`

FoamyGuy, 2025-07-17:
I noticed that `https://httpbin.org/get` seems to succeed a bit more than `https://www.adafruit.com/api/products/6358` and I thought maybe the different sizes of the response could be factoring in.

To test I created a repo with some test files in various sizes https://github.com/FoamyGuy/size_test_files/tree/main/files that can be accessed with their raw github file URLs.

For a little while requesting these files was succeeding. I made it up to 2944.txt file and successfully requested it a few times, as well as many successful requests of smaller files. But kind of suddenly the requests stopped working with the following error:

```
Fetching JSON data from https://github.com/FoamyGuy/size_test_files/raw/refs/heads/main/files/size_1024.txt
Traceback (most recent call last):
  File "/lib/adafruit_requests.py", line 653, in request
  File "/lib/adafruit_requests.py", line 563, in _send_request
  File "/lib/adafruit_requests.py", line 459, in _send
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 124, in send
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 860, in socket_write
ConnectionError: Failed to send 11 bytes (sent 0)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 56, in <module>
  File "/lib/adafruit_requests.py", line 721, in get
  File "/lib/adafruit_requests.py", line 681, in request
OutOfRetries: Repeated socket failures

```

Since that started I haven't gotten any further successful requests. Most of my successful testing was on the Fruit Jam, but after it started failing I tried on the Feather 2350 + Airlift and it is failing for me with the same error.

FoamyGuy, 2025-07-19:
I found a new traceback that I hadn't seen before:
```
Traceback (most recent call last):
  File "code.py", line 69, in <module>
  File "/lib/adafruit_requests.py", line 721, in get
  File "/lib/adafruit_requests.py", line 649, in request
  File "/lib/adafruit_connection_manager.py", line 328, in get_socket
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 60, in getaddrinfo
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 752, in get_host_by_name
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 421, in _send_command_get_response
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 339, in _send_command
TimeoutError: ESP32 timed out on SPI select
```
I've started doing request tests to my server in a loop working its way up thru the sizes of static files. There were several successful requests, and a handful of ETIMEDOUT errors before this new exception occured. Once this exception started it continued happening on every subsequent request.

FoamyGuy, 2025-07-19:
I think that the issue may be in the SPI connection between main MCU and ESP co-processor. I've noticed by watching the logs on the server as requests are being made that there are times when circuitpython reports:
```
Traceback (most recent call last):
  File "/lib/adafruit_requests.py", line 665, in request
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 139, in recv
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 180, in recv_into
OSError: [Errno 116] ETIMEDOUT

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 73, in <module>
  File "/lib/adafruit_requests.py", line 721, in get
  File "/lib/adafruit_requests.py", line 681, in request
OutOfRetries: Repeated socket failures
```
But the server logs seem to indicate successful 200 response from its perspective. 

I added identifying information to the user-agent so I can conveniently correlate requests from client and server logs and I'm noticing that almost every request whether successful or ETIMEDOUT on the circuitpython side is actually making two requests to the server i.e.:
```
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 0"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 0"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 1"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 1"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 2"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 2"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 3"
HTTP/1.1" 200 1024 "-" "Adafruit CircuitPython - 1024 - 3"
```
On the Circuitpython side these only show a single log from my user code indicating that the extra requests must be coming from the internal retry logic from requests or wherever it is.
```
1024 - 0 - success
1024 - 1 - success
1024 - 2 - success
1024 - 3 - E
```
In this slice of the logs there are 3 successes from circuitpython perspective, and one failure with ETIMEDOUT. From the server perspective it was 8 successful `200` responses.

FoamyGuy, 2025-07-22:
I do think it's likely this issue is somehow related to HTTPS. I have a new test script that makes 5 requests against all of the test files from about 1k to about 5k in size. I have found that nina-fw version 3.0 gets 100% success rate when request the `http` URLs, but when I switch to the `https` ones the success rate drops to just under 50%. 

I also tested nina-fw 2.0 with the `https` URLs and see a 100% success rate with that, so this is definitely something new in 3.0.

I believe now that the duplicated requests mentioned in my prior post are unrelated to this specific issue. Those duplicated requests are occurring under 2.0 as well. It is perhaps worth making a separate issue to look into that, maybe somewhere higher level than nina-fw like adafruit_requests, adafruit_connection_manager, or adafruit_esp32spi, but I don't think it's specifically tied to the other issue using version 3.0.

hathach, 2025-07-25:
It seems to the race condition between gpio0updater and command handler when running socket connected() and available(). It works well when commenting out https://github.com/adafruit/nina-fw/blob/36c78d4d4ad9477016ad312464327c01eafac531/main/CommandHandler.cpp#L2510-L2513 . doing more tests

hathach, 2025-07-25:
@dhalbert @FoamyGuy https://github.com/adafruit/nina-fw/pull/72 may help to fix this issue, please give it a try to see if it works

PR #72 (2025-07-28): Fix tls socket race condition
- fix #69 use arduino-esp32 v3.3.0 and build with idf v5.5.
- fix #71 change priority to higher than loop task() to prevent socket connected() and available() race condition which cause bizzare invalid SSL record, the buffer is probably consumed in the wrong task when race occurs. Also correctly give _updateGpio0PinSemaphore in ISR context
- comment out some CONFIG that is mostly default value
closed 2025-07-28 by hathach