adafruit/nina-fw #73: FW 3.0.0 and 3.1.0

rad600, 2025-08-15:
Hello, I have been seeing a consistent issue with 3.0.0 and 3.1.0 while downloading a webpage. This is happening on a esp-32 wifi shield over a Metro M4 and also a Metro M4 Airlift lite. Connecting to a SSL page happens without issue however as the download begins, part of the page loads quickly as expected but then the download slows to a few characters and then stalls. There is no consistency on the byte count before stalling. Trying other pages from various servers exhibits the same issue. For testing purposes I used a modified version of the WifiSSLClient example code.
Reverting to firmware 2.0.0 resolves this issue. 

Thanks

rad600, 2025-08-19:
I used this very basic code to demonstrate the issue. With 3.0.0 and 3.1.0 the serial output showing the web page read initially is ok but after X bytes slows to a sporadic crawl and finally stops before the end of the webpage. FW 2.0.0 is fine in this regard. I also noticed that 2.0.0 is noticeably much faster downloading the web page. Something in 3.0.0 and 3.1.0 seems to be causing a bottleneck.
Thanks for your time.

hathach, 2025-09-09:
thank you for raising the issue and feedback.